ECSHackWeek/impedance.py #272: Error in visualization with looping through and fitting multiple data sets

yika-KYL, 2023-07-11:
I am a new user in impedance py. I used what as instructed in the section of looping through and fitting multiple data sets, in which its visualization part doesn't seem to work out (I could obtain values of fitting circuit parameters) The same script ran in my lab mate's laptop could successfully display all the spectra in one nyquist plot. I wonder if I could get some suggestions on this issue?
[image]

BGerwe, 2023-07-11:
Hi there,

Looks like the arguments passed to `plot_nyquist()` are in the wrong order. Try `plot_nyquist(Z, ax=ax)`

yika-KYL, 2023-07-11:
It works well now! Thank you BGerwe!