ECSHackWeek/impedance.py #276: Very confused about fitting

Alt1139, 2023-08-08:
So I am trying to fit the nyquist plot to the model R0-p(L1,R2)-p(R1,C1). I need these values to confirm similar model used in another plot. Even if i trim the data for 1st quadrant the result is the same. 

frequencies, Z = preprocessing.readCSV('filei.csv')
frequencies = frequencies[1:35]
Z = Z[1:35]
circuit = 'R0-p(L1,R2)-p(R1,C1)'
initial_guess = [12e-3, 2e-7, 1e-3, 1.23e-3, 0.2e-1]
circuit = CustomCircuit(initial_guess=initial_guess, circuit=circuit)
circuit.fit(frequencies, Z)
Z_fit = circuit.predict(frequencies)

[image]

[image]

Yugal0, 2023-08-09:
Where do you obtain this data (filei.csv') from? 
1. Is it from some experiments?
2. Or, it is some synthetic data, that you, yourself have generated by simulating the exact same circuit (that you are trying to fit to the data (R0-p(L1,R2)-p(R1,C1))) or by simulating any other similar circuit.?

Please answer these, I may be able to help, after knowing this.

I think maybe the possible problems can be,
1.If your data is coming from an experiment, then the circuit you have proposed is not sufficient to represent the 
   experimental data.
2. Sometimes the fitting process is very much influenced by the initial guess that you provide, so you should also check the initial guess values (If your data is coming from the simulation of another circuit, for which you obviously know the value of these parameters). Try putting those exact values as the initial guess.

Let me know if there is any progress or further problems. I will be happy to help.

Alt1139, 2023-08-16:
The data is from an experiment of a Normal NMC LIB. I was reading a lot of literature in which it mentions that the semicircle part of the plot mainly represents the -p(R1,C1) and I am very much interested in this value. For reference:
[image]

I even tried to only fit the semi circle area and no luck. I also read that ZARC Frequency Domain Fitting is required to fit this region of the battery.

Yugal0, 2023-08-19:
I think if you want to fit a circuit to the complete experimental data that is there in your paper, 
1. you should not take resistance in parallel with the inductor (you have done this in your code (above))
2. and also replace the capacitor with a Constant Phase Element (CPE) (say with alpha=0.9)
3. and try taking a CPE instead of the Warburg element (that is there in the Paper), we also recently faced this problem with the Warburg element, the CPE can approximate the Warburg element atleast in some frequency range.