ECSHackWeek/impedance.py #279: Initial_guess

merung1203, 2023-09-05:
Hello

How do I set the initial value when fitting a circuit?

I am using the intercept for R0, but I would like to know how to set the rest.
I would like to know why these initial values ​​are entered in the example code below.

example code)
from impedance.models.circuits import CustomCircuit

circuit = 'R0-p(R1,C1)-p(R2-Wo1,C2)'
initial_guess = [.01, .01, 100, .01, .05, 100, 1]

circuit = CustomCircuit(circuit, initial_guess=initial_guess)

If the initial value is set incorrectly, an error occurs.  I wonder if it is possible to set the initial value as a range.
Thank you.

jianzuo, 2023-09-18:
Hi, did you manage to solve the initial value issue? I have a similar issue, 
when the initial value is not properly decided, the fitted curve is inaccurate.

quiksilver10152, 2024-05-06:
Try using the global optimization algorithm setting. 
circuit.fit(frequencies, Z, global_opt = True)

BGerwe, 2026-02-27:
Sorry we never got back to you about the issue. Hopefully you were able to resolve it, but reopen this if you're still having trouble.

closed 2026-02-27 by BGerwe