ECSHackWeek/impedance.py #282: A worked example of .nox Metrohm file reading and pre-processing

AlexanderWilson1993, 2023-10-25:
Hi there,

I have been looking through the documentation and have been unable to find an actual example where the file is coming from a .nox Metrohm file type. I have tried to write the code on my own and have had no luck. If you could add a worked example that would be great!

This would be helpful for other users with different file types other than .csv.

Best wishes,

Alex

AlexanderWilson1993, 2023-11-03:
Getting this error: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 5: invalid start byte

When I try to open an EIS .nox file.

mdmurbach, 2023-11-24:
Hi @AlexanderWilson1993, do you have an example file you can share? It seems like the file is not utf-8 encoded so the code will need to read in the file as bytes (and then decoded them according to the proper format which we might have to guess). Unfortunately, it seems like a lot of the instrument manufacturers do this to make it difficult for you to use anything but their own proprietary software for analysis :(