ECSHackWeek/impedance.py #283: high frequency fitting problem

z1q0831, 2023-10-25:
hallo, i have a question about high freq fitting. I have used RL to fit but seems does not work.
the fitting of positive Zimg is not possible
can anyone maybe know why?plz
[image]
[image]

z1q0831, 2023-10-26:
@Yugal0

Yugal0, 2023-10-26:
Although I will give it a try myself tomorrow, but for now I think you can try increasing the value of inductance in the initial guess.

z1q0831, 2023-10-26:
> Although I will give it a try myself tomorrow, but for now I think you can try increasing the value of inductance in the initial guess.

thank you for the suggestion. I have tried to increase the value of inductor, but still can not get the the fitting of positive Zimag. the csv i use is from the [exampleData.csv](https://impedancepy.readthedocs.io/en/latest/_downloads/320671c0bb666e4d6ac487c9d7ff1679/exampleData.csv).

Yugal0, 2023-10-26:
> thank you for the suggestion. I have tried to increase the value of inductor, but still can not get the the fitting of positive Zimag. the csv i use is from the [...]

Ok I will try fitting myself.(tommorow)

Yugal0, 2023-10-27:
What I got in the first try:

`from impedance.models.circuits import CustomCircuit`
`import numpy as np`
`import pandas as pd`
`import matplotlib.pyplot as plt`

`initial_guess=[0.01,0.01,100,0.01,0.05,100,1,1,0.1]`
`c=CustomCircuit("R0-p(R1,C1)-p(R2-Wo1,C2)-p(L1,R3)",initial_guess=initial_guess)`
`df=pd.read_csv("../exampleData.csv")`
`gthbdata=df.to_numpy()`
`datanew=np.transpose(gthbdata)`
`freq=datanew[0]`
`z_data=datanew[1]+1j*datanew[2]`
`plt.plot(np.real(z_data),-np.imag(z_data))`
[image]

`c.fit(frequencies=freq,impedance=z_data)`
`print(c)`
`z_fit=c.predict(frequencies=freq)`
`plt.plot(np.real(z_fit),-np.imag(z_fit))`
`plt.plot(np.real(z_data),-np.imag(z_data))`
[image]
[image]

z1q0831, 2023-10-27:
> What I got in the first try:
>
> `from impedance.models.circuits import CustomCircuit` `import numpy as np` `import pandas as pd` `import matplotlib.pyplot as plt` [...]

thank you for your try, so as I see in the last picture, the fitting line also can not reach the positive Zimag?

Yugal0, 2023-10-27:
> thank you for your try, so as I see in the last picture, the fitting line also can not reach the positive Zimag?

Yes, same problem as yours happening.
But I made another mistake, the plots are not 1:1 in aspect ratio.
And also, seems like the fit you are getting and what I am getting are different, why so? what method are you using for fitting?
YOUR FIT:
[image]
MY FIT: (now its 1:1)
[image]

z1q0831, 2023-10-27:
> Yes, same problem as yours happening. But I made another mistake, the plots are not 1:1 in aspect ratio. And also, seems like the fit you are getting and what I am getting are different, why so? [...]

do you have the same initial guess of mine? Maybe i have add another (RC) component, because (RC) is responsible for semicircle part.

Yugal0, 2023-10-27:
yes I put the exact same initial guess (and the same elements) as you showed in the code in your very first snapshot in the question. 

Actually just now, what I found is if I take L=100 as the initial guess the I am getting a pretty good fit 
MY FIT NOW:
[image]
In your snapshot it was L=1, with that I was getting not a good fit (as shown in my previous comments) 
MY FIT BEFORE:
[image]

But still how you getting good fit with L=1 as initial guess?
YOUR FIT as shown in your first comment:
[image]

z1q0831, 2023-10-27:
> yes I put the exact same initial guess (and the same elements) as you showed in the code in your very first snapshot in the question. YOUR SNAPSHOT of CODE: [image]
> 
> Actually just now, what I found is if I take L=100 as the initial guess the I am getting a pretty good fit MY FIT NOW: [image] In your snapshot it was L=1, with that I was getting not a good fit (as shown in my previous comments) MY FIT BEFORE: [image]
> 
> But still how you getting good fit with L=1 as initial guess? YOUR FIT as shown in your first comment: [image]

Maybe I have changed the value of L , but the problem still remains. The high frequency fitting can not be reached.

Yugal0, 2023-10-27:
Yeah, that is the main problem. I will let you know here, If I get any solutions for this.

z1q0831, 2023-10-27:
> Yeah, that is the main problem. I will let you know here, If I get any solutions for this.

But totally thank you for replying me! If I find any solutions, I will post here