ECSHackWeek/impedance.py #284: Accuracy of data in .par file about VersaStudio

Earnest-pyja, 2023-11-10:
Hello,

I have a problem in dealing with the data in .par file. Because i found the Zre and Zim data in .par file in different from the data exported with VersaStudio. 
The manual of VersaStudio also points out this issue:"The EIS data within the raw data file is un-calibrated and should not be used for data analysis. Only EIS data that is copied and pasted from within VersaStudio should be used for any data exportation and analysis. "
I read the code of impedance.py, it seems raw data of .par file is not processed.
So I want to know how to deal with it, or I missed the code dealing with this issue.

Thanks.

mdmurbach, 2023-11-24:
Unfortunately, if the data in the .par file is not meant to be used for data analysis, then I don't think that impedance.py will be able to be used (unless you know what processing VersaStudio is applying and we can add it). You can see [here](https://github.com/ECSHackWeek/impedance.py/blob/558b51c6f97166e3fc5823886993e4b6f2c4e914/impedance/preprocessing.py#L218), that the readVersaStudio is just reading in the Segment value directly. @BGerwe have you seen this issue at all? Should we add a disclaimer to the readVersaStudio function? 

@Earnest-pyja do you have a link to the manual/section that you can share so we can add a reference?

BGerwe, 2023-11-26:
Unfortunately, I'm not very familiar with this issue. I briefly worked with data in .par format, but my collaborators collected the data so I never really interacted with their software

Earnest-pyja, 2023-11-28:
> Unfortunately, if the data in the .par file is not meant to be used for data analysis, then I don't think that impedance.py will be able to be used (unless you know what processing VersaStudio is [...]

I found it in this website (page 104--Appendix 3): https://manualzz.com/doc/28282919/versastudio---princeton-applied-research.
Actually, this sentence is from <VersaStudio Software Manual>, you could also find it in official manual.

BGerwe, 2026-02-27:
Given this, we can either add this disclaimer into the [readVersaStudio](https://github.com/ECSHackWeek/impedance.py/blob/558b51c6f97166e3fc5823886993e4b6f2c4e914/impedance/preprocessing.py#L218) or remove the function entirely. I'm inclined toward the former, but curious if that potentially creates more trouble for folks who may be unwittingly analyzing poor data.