ECSHackWeek/impedance.py #289: Adding Distribution of relaxation times feature

kevinsmia1939, 2024-03-17:
So I see a bit of discussion on adding distribution of relaxation times (DRT) feature, but that would be in a far future since it would take quite some time to implement it.

Recently I stumble upon pyDRTtools (https://github.com/ciuccislab/pyDRTtools) which is the Python implementation of DRT from the same author as DRTtools.
I think we can take a look at it. I'm looking into it but my knowledge on DRT is not enough yet.

nicmazza, 2024-03-26:
A couple more of interesting packages are:
-dearEIS, which is fairly complete and includes a GUI (https://vyrjana.github.io/DearEIS/guide_drt.html)
-bayes-drt, which apparently requires less tuning of the parameters https://github.com/jdhuang-csm/bayes-drt.

mdmurbach, 2024-04-30:
Thanks for the links. I'd be happy to help review a PR/figure out how to integrate with those DRT tools (maybe it's as simple as writing a translator between the impedance.py circuit string definitions and the other package inputs)